kroshu/kuka_robot_descriptions #10: From to support to description

Groebehavn, 2023-03-03:
Packages are named "support", contradicting the naming convention of "description". One should get over the repository, check conventions, rename, find if there are any ROS1 leftovers, etc.

gavanderhoorn, 2023-03-05:
Seeing as at least one of the packages referred to was copied from `ros-industrial/kuka_experimental`, I figured I'd provide some history / context to the naming. As it's all OSS, you're free to do whatever you prefer of course, but there's a reason for the naming.

See [wiki.ros.org: Working with ROS-Industrial Robot Support Packages](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages), specifically the *Overview* section. Summarising: it's somewhat *awkward* to have `_description` packages with multiple variants of the same series of robots, as is often the case with industrial robots. To make it more generic, the suffix `_support` was chosen.

---

Having written that: would it perhaps be an idea to try and reduce duplication of effort and files? There are other people/groups also porting `_support` packages to ROS 2, and it seems like you're (planning on) doing the same thing.

As `.xacro`s are pretty driver-agnostic (if done right), there doesn't seem to be any particular advantage in (re)doing the same work.

Would coordinating efforts be an option you'd be interested in?

gavanderhoorn, 2023-03-06:
> we are developing a tool that is able to generate these packages automatically from the exact robot model data - both for ROS1 and ROS2.

Ok.

I can't say more on this public forum, but if you contact me via e-mail, I could potentially avoid more duplication of effort.

closed 2023-07-14 by Svastits