kroshu/kuka_robot_descriptions #27: URDF problems with ROS humble

terkaa, 2023-10-07:
Hi,

  For some reason ROS humble does not read URDF files correctly. The robot has only three first joints and those are incorrectly placed. Should I change to another distro, or is there other way to solve the issue.   

[image]

Tero

Svastits, 2023-10-09:
Can you provide some additional details? Which robot are you trying to load? How (using which launch file) are you trying to load?
We have tested every urdf with humble, I would not say the ros distro should be an issue

terkaa, 2023-10-09:
Launch command is: 

ros2 launch kuka_rsi_hw_interface startup_with_rviz.launch.py

[INFO] [launch]: All log files can be found below /home/tero/.ros/log/2023-10-09-16-15-50-270526-Aorus-9757
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [control_node-1]: process started with pid [9770]
[INFO] [robot_manager_node-2]: process started with pid [9772]
[INFO] [robot_state_publisher-3]: process started with pid [9774]
[INFO] [spawner-4]: process started with pid [9776]
[INFO] [spawner-5]: process started with pid [9778]
[INFO] [rviz2-6]: process started with pid [9780]
[robot_state_publisher-3] [INFO] [1696857350.547958344] [robot_state_publisher]: got segment base
[robot_state_publisher-3] [INFO] [1696857350.548012034] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1696857350.548016921] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-3] [INFO] [1696857350.548019973] [robot_state_publisher]: got segment flange
[robot_state_publisher-3] [INFO] [1696857350.548022477] [robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1696857350.548024966] [robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1696857350.548027321] [robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1696857350.548029585] [robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1696857350.548031982] [robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1696857350.548034292] [robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1696857350.548036630] [robot_state_publisher]: got segment tool0
[control_node-1] [INFO] [1696857350.549381920] [resource_manager]: Loading hardware 'kr6_r700_sixx' 
[control_node-1] [INFO] [1696857350.549851920] [resource_manager]: Initialize hardware 'kr6_r700_sixx' 
[control_node-1] [INFO] [1696857350.549982678] [KukaRSIHardwareInterface]: IP of client machine: 0.0.0.0:59152
[control_node-1] [INFO] [1696857350.549990153] [resource_manager]: Successful initialization of hardware 'kr6_r700_sixx'
[control_node-1] [INFO] [1696857350.550062323] [resource_manager]: Hardware Component with name '' does not exists
[control_node-1] [ERROR] [1696857350.552375285] [controller_manager]: setscheduler error
[control_node-1] [ERROR] [1696857350.552440946] [controller_manager]: Inappropriate ioctl for device
[control_node-1] [WARN] [1696857350.552445352] [controller_manager]: You can use the driver but scheduler priority was not set
[control_node-1] [INFO] [1696857350.710618588] [controller_manager]: Loading controller 'joint_state_broadcaster'
[control_node-1] [INFO] [1696857350.716869806] [controller_manager]: Loading controller 'joint_trajectory_controller'
[control_node-1] [WARN] [1696857350.720192493] [joint_trajectory_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[spawner-4] [INFO] [1696857350.724537356] [spawner_joint_state_broadcaster]: [94mLoaded [1mjoint_state_broadcaster[0m
[control_node-1] [INFO] [1696857350.725130404] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[control_node-1] [INFO] [1696857350.725178136] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-5] [INFO] [1696857350.736558994] [spawner_joint_trajectory_controller]: [94mLoaded [1mjoint_trajectory_controller[0m
[spawner-5] [INFO] [1696857350.740279130] [spawner_joint_trajectory_controller]: [96mLoaded parameters file "/home/tero/kukaros_ws/install/kuka_rsi_hw_interface/share/kuka_rsi_hw_interface/config/joint_trajectory_controller_config.yaml" for [1mjoint_trajectory_controller[0m
[spawner-5] [INFO] [1696857350.740623645] [spawner_joint_trajectory_controller]: Loaded /home/tero/kukaros_ws/install/kuka_rsi_hw_interface/share/kuka_rsi_hw_interface/config/joint_trajectory_controller_config.yaml into joint_trajectory_controller
[control_node-1] [INFO] [1696857350.741180524] [controller_manager]: Configuring controller 'joint_trajectory_controller'
[control_node-1] [INFO] [1696857350.742570928] [joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[control_node-1] [INFO] [1696857350.742594408] [joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position].
[control_node-1] [INFO] [1696857350.742606373] [joint_trajectory_controller]: Using 'splines' interpolation method.
[control_node-1] [INFO] [1696857350.742986456] [joint_trajectory_controller]: Controller state will be published at 50.00 Hz.
[control_node-1] [INFO] [1696857350.743569100] [joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[INFO] [spawner-4]: process has finished cleanly [pid 9776]
[INFO] [spawner-5]: process has finished cleanly [pid 9778]
[INFO] [rviz2-6]: process has finished cleanly [pid 9780]
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[control_node-1] [INFO] [1696857354.648335506] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_manager_node-2]: process has finished cleanly [pid 9772]
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 9774]
[INFO] [control_node-1]: process has finished cleanly [pid 9770]
[robot_manager_node-2] [INFO] [1696857354.648340253] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-3] [INFO] [1696857354.648368700] [rclcpp]: signal_handler(signum=2)

Svastits, 2023-10-10:
If you would like only to visualize the robot modell in rviz, you should use the launch files available in the kuka_simulators repo under package_name/launch

The launch file you started also starts the hardware interface and some ros2_controllers alongside rviz, and therefore the robot model will not be visible in rviz, until you have a valid state from a robot. So you can either connect to a real robot, or launch the basic simulator (ros2 launch kuka_rsi_simulator kuka_rsi_simulator_launch.py)
Additionally, the driver must be configured and activated, as a lifecycle node manages hardware interface activation:
ros2 lifecycle set robot_manager configure
ros2 lifecycle set robot_manager activate

closed 2023-11-09 by Svastits