kroshu/kuka_robot_descriptions #41: Robot model fails to load properly on Humble

Tejas-Shastha, 2023-12-14:
On Humble, when I build the main branch and launch the provided KR moveit launch file with fake controllers, I notice that the robot model is not properly loaded with the following message in the terminal:

```
WARNING:root:Cannot infer URDF from `/home/<user>/workspaces/kuka_ws/install/kuka_kr_moveit_config/share/kuka_kr_moveit_config`. -- using config/kuka_kr.urdf
WARNING:root:Cannot infer SRDF from `/home/<user>/workspaces/kuka_ws/install/kuka_kr_moveit_config/share/kuka_kr_moveit_config`. -- using config/kuka_kr.srdf

```
In rviz I can see that TF for links1-6, tool and flange are also not available.

Svastits, 2023-12-14:
These warnings are issued by the moveit setup assistant, which expects a different folder structure from the convention in ROS-Industrial also used by us. This does not influence the loading of the robot model.

The reason for the not appearing robot meshes is that currently you have to activate the robot manager node even for fake hardware, as [described in the readme](https://github.com/kroshu/kuka_robot_descriptions#:~:text=This%20will%20launch,examples\)/iiqka_moveit_example%20package.)). This behaviour will change soon, I would suggest to use [this branch](https://github.com/kroshu/kuka_robot_descriptions/tree/feature/kr10_r1100), as the scheduling is also optimized there.

Tejas-Shastha, 2023-12-15:
> you have to activate the robot manager node even for fake hardware, as described in the readme

That was indeed the case, as I unfortunately skipped the very last paragraph of the readme in my haste :) 

Thank you for both the tips!

closed 2023-12-15 by Tejas-Shastha