kroshu/kuka_robot_descriptions #51: Moveit interactive marker (blue sphere) not available

prasuchit, 2024-04-09:
Hi,

When I run the moveit examples for lbr iisy 3 from this package, I don't see the blue sphere around the end effector that is usually used to move the robot around freely and set a goal state for Moveit.

How do I enable this feature?

@Svastits

Svastits, 2024-04-10:
Unfortunately I also did not manage to conjure up the interactive markers, they were also not visible for me, when I ran the `moveit2_tutorials`...
I will try to look into this more deeply sometime

neerja123-gupta, 2024-06-07:
Hi, were you able to resolve this issue? I'm experiencing something similar. I can't see the blue sphere and I'm unable to move the robot in Cartesian space using RViz. I haven't tried using the C++ or Python interface yet.
[image]

Thanks.

prasuchit, 2024-06-07:
@neerja123-gupta no, it hasn't been resolved yet, but @Svastits is aware of it and is planning to resolve it soon. 

You might want to be aware of these other issues I've come across so far as well:

[Comment] (https://github.com/kroshu/kuka_drivers/issues/167#issuecomment-2147863331)

neerja123-gupta, 2024-06-07:
@prasuchit  ok thanks a lot for the information.

prasuchit, 2024-07-30:
@neerja123-gupta, check out [this comment](https://github.com/kroshu/kuka_drivers/issues/167#issuecomment-2258724498) for my solution to this problem! Hope that helps. :)

neerja123-gupta, 2024-08-14:
@prasuchit oh thats great. Thank you so much @prasuchit .

closed 2024-08-26 by Svastits